openvizsla/ov_ftdi #8: Where's the license?

micolous, 2018-03-03:
Looking over this repository, I cannot find any license.  Please add one!

https://choosealicense.com/no-permission/

Goddard, 2018-03-10:
The guy that made this hardware and software passed away.

micolous, 2018-03-11:
Hmm... this is difficult then.

I happened to borrow some of this hardware to debug some USB issues I'm having.  I'm trying to "finish" the Wireshark module. There's some code in sigrok that implements a state machine that allows transforming USB link-layer into the same sort of USB captures that you get out of Linux usbmon. That would be good enough.

However, in order to incorporate the two together would violate the license of sigrok -- as that is GPL, and OV is "???".

The only way I see around that is to entirely reimplement the host software side of things.  This is difficult for me as there is also little documentation about how the host-device communication works. I don't feel comfortable working from the existing source release when the license is undefined.

Even if I find some other documentation here that would point to a license from the original author, there are also other contributors to the project, and _their_ intent is also unclear.

I could write something up that solves my problems, but I would not be able to release it.

I'm just mindful of getting into too much yak shaving. ;)

Goddard, 2018-03-11:
Well the designs of the hardware are under creative commons.  As for the code, most people deal with instances like that by just having an automated downloader and if a third-party provides the binaries you can even auto-download those and the user clicks the button which results in no infringement on a license.

somehdlguy, 2018-03-11:
Hey all, re license. Everything I wrote (parts of the host python, comms and the FPGA HDL) may be considered to be under 2-clause BSD. I'm speaking with some of the others that wrote large chunks of code (tmbinc, marcan et al) and we'll agree on a license and put a license statement up.

It'll be something open source, just not sure on exactly which one we'll end up with. (probably GPL-2-or-later, or BSD would be my guess)

somehdlguy, 2018-03-11:
Also, I can talk about

tmbinc, 2018-03-11:
For what I wrote:

 - Wireshark plugin is probably derived work, so it'll be GPLv2. I'll add a LICENSE immediately.
 - For any HDL code that I wrote I'm fine with 2-clause BSD. Once we agreed, we'll put a LICENSE file there.

somehdlguy, 2018-03-11:
I spoke to marcan, he's OK with 2-clause BSD for everything he touched. You wrote / extended a bunch of the host side python code as well (I assume you are OK with 2-clause BSD on that?).

In any case, between the 3 of us, we wrote the vast majority of the host-side and device side code. Ben mostly contributed bugfixes to the host-side code, his primary responsibility was the hardware (which as is noted above, he marked CC-BY-SA).

I think we are good to go with 2-clause BSD then, with an exception for the wireshark derived code which would be GPLv2.

tmbinc, 2018-03-11:
Sounds good!

closed 2018-03-11 by tmbinc

micolous, 2018-03-13:
Thanks for clearing this up!