openvizsla/ov_ftdi #9: Hardware for project

tbutton, 2018-05-12:
All versions of Openvizsla out of stock. Could you help please suggest another board for this project. Thank you!

laf0rge, 2018-06-07:
I'm not in the US but would be interested in buying any of the "old" OpenViszla boards.  In fact, I'm even interested in building some more, as needed (I regularly design + build all kinds of electronics).

The big question to me is what happened to the project, given the huge kickstarter success 4 years ago, and the statement from @tmbinc that hardware/gateware is "done".  As a regular wireshark contributor, I'm not so worried about having to improve the visualization side on the host.  Also, these days sigrok has USB trace decoding/displaying capabilities, too.

laf0rge, 2018-06-07:
thanks for your feedback.  I think I would be interested in playing with one of the units for some time in order to see if and how to move it ahead.  Making the hardware available is not a problem for me, we already make available several circuit boards from the @osmocom community and distribute it over the sysmocom webshop, such as the Osmcoom SIMtrace, which is a protocol tracer for SIM cards (or actually any smart cards).

Please note that I'm not primarily interested in *selling* something, but simply in enabling anyone to easily get their hands on a related board if they'd like to play with it.

To me, this project looks a bit like a "hidden gem".  A lot of work was done years ago, several people claim that the it's fundamentally working.  There doesn't appear to be any FOSS / OpenHardware alternative, and for many use cases sniffing FS/HS USB is also still more than sufficient.

It would be great if I could somehow get my hands on a "known working" board first to avoid having to spend time on building a board from scratch and doign board-bringup, testing, etc. on a device that I'm not familiar with yet...

somehdlguy, 2018-06-07:
@laf0rge: I'll follow up with you directly via email or IRC regarding availability of a board for development purposes. 

Separately, I'll see whether the tree still builds with current versions of dependencies, and if necessary, fix the build.

cyphunk, 2018-08-20:
@laf0rge I have one from the original kickstarted I'd happily loan you. I know it worked a few years ago.

laf0rge, 2021-05-31:
In case it hasn't been noticed meanwhile, there are two sources for OpenVizsla that have appeared manwhile, see the links in the README.md at https://github.com/openvizsla/ov_ftdi#obtaining-hardware

closed 2021-05-31 by tmbinc