openvizsla/ov_ftdi #10: Migrate FPGA firmware to the latest Migen

matwey, 2018-07-06:
Well, if somebody have a branch with support of the latest Migen, please put a note here.

somehdlguy, 2018-07-06:
I have an in-progress port to the latest migen. It currently generates a bitstream that isn't fully tested yet, and I still need to fix some of the sim tests. I should be able to finish testing it in the next few days at which point  I'll push the branch.

matwey, 2018-07-06:
@somehdlguy really nice to hear that. Do you have write access to this repo?

somehdlguy, 2018-07-06:
I do, and I'll push it here once I have it tested. I'll close this issue at that time so you get a notification.

matwey, 2018-07-06:
Thank you very much.

matwey, 2018-08-04:
What do I do wrong? I use migen and misoc from master.

```
> make
python3 build.py build_dir build build_name ov3
. /opt/Xilinx/14.7/ISE_DS/common/.settings64.sh /opt/Xilinx/14.7/ISE_DS/common
. /opt/Xilinx/14.7/ISE_DS/EDK/.settings64.sh /opt/Xilinx/14.7/ISE_DS/EDK
. /opt/Xilinx/14.7/ISE_DS/PlanAhead/.settings64.sh /opt/Xilinx/14.7/ISE_DS/PlanAhead
. /opt/Xilinx/14.7/ISE_DS/ISE/.settings64.sh /opt/Xilinx/14.7/ISE_DS/ISE
Release 14.7 - xst P.20131013 (lin64)
Copyright (c) 1995-2013 Xilinx, Inc.  All rights reserved.
--> 
WARNING:Xst:1583 - You are using an internal switch '-use_new_parser'.

TABLE OF CONTENTS
  1) Synthesis Options Summary
  2) HDL Parsing
  3) HDL Elaboration
  4) HDL Synthesis
       4.1) HDL Synthesis Report
  5) Advanced HDL Synthesis
       5.1) Advanced HDL Synthesis Report
  6) Low Level Synthesis
  7) Partition Report
  8) Design Summary
       8.1) Primitive and Black Box Usage
       8.2) Device utilization summary
       8.3) Partition Resource Summary
       8.4) Timing Report
            8.4.1) Clock Information
            8.4.2) Asynchronous Control Signals Information
            8.4.3) Timing Summary
            8.4.4) Timing Details
            8.4.5) Cross Clock Domains Report

=========================================================================
*                      Synthesis Options Summary                        *
=========================================================================
---- Source Parameters
Input File Name                    : "top.prj"
Input Format                       : MIXED

---- Target Parameters
Output File Name                   : "top.ngc"
Target Device                      : xc6slx9-tqg144-3

---- Source Options
Top Module Name                    : top
Use New Parser                     : yes
Automatic Register Balancing       : yes

---- General Options
Optimization Goal                  : SPEED

=========================================================================

WARNING:Xst:29 - Optimization Effort not specified
=========================================================================

=========================================================================
*                          HDL Parsing                                  *
=========================================================================
Analyzing Verilog file "/home/matwey/lab/ov_ftdi/software/fpga/ov3/build/top.v" into library work
Parsing module <top>.
ERROR:HDLCompiler:1660 - "/home/matwey/lab/ov_ftdi/software/fpga/ov3/build/top.v" Line 4518: Procedural assignment to a non-register overflowinserter_re is not permitted, left-hand side should be reg/integer/time/genvar
ERROR:HDLCompiler:1660 - "/home/matwey/lab/ov_ftdi/software/fpga/ov3/build/top.v" Line 4518: Procedural assignment to a non-register overflowinserter_re is not permitted, left-hand side should be reg/integer/time/genvar
ERROR:HDLCompiler:1660 - "/home/matwey/lab/ov_ftdi/software/fpga/ov3/build/top.v" Line 4881: Procedural assignment to a non-register sdram_sink_ptr_read_re is not permitted, left-hand side should be reg/integer/time/genvar
ERROR:HDLCompiler:1660 - "/home/matwey/lab/ov_ftdi/software/fpga/ov3/build/top.v" Line 4881: Procedural assignment to a non-register sdram_sink_ptr_read_re is not permitted, left-hand side should be reg/integer/time/genvar
ERROR:HDLCompiler:1660 - "/home/matwey/lab/ov_ftdi/software/fpga/ov3/build/top.v" Line 4921: Procedural assignment to a non-register sdram_sink_debug_ctl_re is not permitted, left-hand side should be reg/integer/time/genvar
ERROR:HDLCompiler:1660 - "/home/matwey/lab/ov_ftdi/software/fpga/ov3/build/top.v" Line 4921: Procedural assignment to a non-register sdram_sink_debug_ctl_re is not permitted, left-hand side should be reg/integer/time/genvar
ERROR:HDLCompiler:1660 - "/home/matwey/lab/ov_ftdi/software/fpga/ov3/build/top.v" Line 5063: Procedural assignment to a non-register sdram_sink_ptr_read_re is not permitted, left-hand side should be reg/integer/time/genvar
ERROR:HDLCompiler:1660 - "/home/matwey/lab/ov_ftdi/software/fpga/ov3/build/top.v" Line 5063: Procedural assignment to a non-register sdram_sink_ptr_read_re is not permitted, left-hand side should be reg/integer/time/genvar
ERROR:HDLCompiler:1660 - "/home/matwey/lab/ov_ftdi/software/fpga/ov3/build/top.v" Line 5076: Procedural assignment to a non-register sdram_sink_debug_ctl_re is not permitted, left-hand side should be reg/integer/time/genvar
ERROR:HDLCompiler:1660 - "/home/matwey/lab/ov_ftdi/software/fpga/ov3/build/top.v" Line 5076: Procedural assignment to a non-register sdram_sink_debug_ctl_re is not permitted, left-hand side should be reg/integer/time/genvar
ERROR:HDLCompiler:1660 - "/home/matwey/lab/ov_ftdi/software/fpga/ov3/build/top.v" Line 5107: Procedural assignment to a non-register overflowinserter_re is not permitted, left-hand side should be reg/integer/time/genvar
ERROR:HDLCompiler:1660 - "/home/matwey/lab/ov_ftdi/software/fpga/ov3/build/top.v" Line 5107: Procedural assignment to a non-register overflowinserter_re is not permitted, left-hand side should be reg/integer/time/genvar
ERROR:HDLCompiler:1660 - "/home/matwey/lab/ov_ftdi/software/fpga/ov3/build/top.v" Line 5232: Procedural assignment to a non-register data_out_source_stb is not permitted, left-hand side should be reg/integer/time/genvar
[... cut ...] ime/genvar
ERROR:HDLCompiler:1660 - "/home/matwey/lab/ov_ftdi/software/fpga/ov3/build/top.v" Line 5235: Procedural assignment to a non-register data_out_source_payload_rxcmd is not permitted, left-hand side should be reg/integer/time/genvar
ERROR:HDLCompiler:1660 - "/home/matwey/lab/ov_ftdi/software/fpga/ov3/build/top.v" Line 5239: Procedural assignment to a non-register data_out_source_payload_d is not permitted, left-hand side should be reg/integer/time/genvar
Sorry, too many errors..
--> 
```

matwey, 2018-08-04:
There is no branch `new_migen` in `tmbinc/ov_ftdi`. I've tried `new_migen` from `openvizsla/ov_ftdi`.

matwey, 2018-08-04:
I am sorry, I've found correct branch. Thank you.