openvizsla/ov_ftdi #17: Output information from OV

marekmatw, 2018-09-11:
Hello everyone. I've been testing an OpenVizsla v3.2 device and I have a couple of problems. I'm trying to analyze a pendrive [https://stat-s9.ms-online.pl/media/cache/gallery/rc/b5o50r65/bdk/78/786367_0.jpg](url) and USB-UART converter.

I'm using software from the `master` branch. All of the tests are ok so the board is assembled correctly (I think)

But when I do sniffing, I receive such information, but nothing stems from it. 

```
[        ]  60.145390 d=  0.000003 [ 69.4 + 89.817] [  3] IN   : 12.1 
[        ]  60.145393 d=  0.000003 [ 69.4 + 92.467] [  3] IN   : 12.1 
[        ]  60.145395 d=  0.000003 [ 69.4 + 95.117] [  3] IN   : 12.1 
[        ]  60.145405 d=  0.000009 [ 69.4 +104.483] [  1] ACK 
[        ]  60.145405 d=  0.000000 [ 69.4 +104.967] [  3] IN   : 12.1 
[        ]  60.145408 d=  0.000003 [ 69.4 +107.617] [  3] IN   : 12.1 
[        ]  60.145410 d=  0.000003 [ 69.4 +110.267] [  3] IN   : 12.1 
[        ]  60.145426 d=  0.000015 [ 69.5 +  0.583] [  3] IN   : 12.1 
[        ]  60.145431 d=  0.000005 [ 69.5 +  5.900] [  3] IN   : 12.1 
[        ]  60.145434 d=  0.000003 [ 69.5 +  8.550] [  3] IN   : 12.1 
[        ]  60.145436 d=  0.000003 [ 69.5 + 11.200] [  3] IN   : 12.1 
```
I expected the output to be like in the manual for `HS sniffing` for a USB flash drive
```
$ ./ovctl.py -p ov3.fwpkg sniff hs
[   F    ]   0.267814 d=  0.267814 [   .0 +267814.250] [  0]  
[        ]   1.123487 d=  0.855673 [255.7 +  0.600] [  3] IN   : 2.1 
[        ]   1.123488 d=  0.000001 [255.7 +  1.250] [  1] ACK 
[        ]   1.224924 d=  0.101436 [101.2 + 54.550] [  3] SETUP: 2.0 
[        ]   1.224924 d=  0.000000 [101.2 + 54.967] [ 11] DATA0: a3 00 00 00 02 00 04 00 f6 e1 
[        ]   1.224927 d=  0.000003 [101.2 + 57.933] [  3] IN   : 2.0 
[        ]   1.224928 d=  0.000001 [101.2 + 58.633] [  1] ACK 
[        ]   1.224930 d=  0.000002 [101.2 + 60.800] [  3] OUT  : 2.0 
[        ]   1.224930 d=  0.000000 [101.2 + 61.217] [  3] DATA1: 00 00 
[        ]   1.225175 d=  0.000245 [101.4 + 55.900] [  3] SETUP: 2.0 
[        ]   1.225175 d=  0.000000 [101.4 + 56.317] [ 11] DATA0: 23 01 14 00 02 00 00 00 ef 95 
...
# 0000000: 2203 4400 6100 7400 6100 5400 7200 6100  ".D.a.t.a.T.r.a.
# 0000010: 7600 6500 6c00 6500 7200 2000 3200 2e00  v.e.l.e.r. .2...
# 0000020: 3000 ce3a                                0..:
[        ]   1.242314 d=  0.000000 [118.5 + 68.383] [ 37] DATA1: 22 03 44 00 61 00 74 00 61 00 54 00 72 00 61 00 76 00 65 00 6c 00 65 00 72 00 20 00 32 00 2e 00 30 00 ce 3a 
```
When I use a card reader, I obtain
```
Unmatched byte a5 - discarding
Unmatched byte 49 - discarding
Unmatched byte 64 - discarding
0 / 16777216 (0.00 % utilization) 0 kB | 0 overflow, 00c3537c total | R0000012c W0000012c
0 overflow, 02c433e5 total
0 / 16777216 (0.00 % utilization) 0 kB | 0 overflow, 09a1e6e3 total | R0000012c W0000012c
0 overflow, 0ba2608a total
0 / 16777216 (0.00 % utilization) 0 kB | 0 overflow, 12801e45 total | R0000012c W0000012c
```
Or maybe I don't how it means to be used. Can anyone explain it?

marcan, 2018-09-11:
Your first log looks fine. That's just idle traffic on the bus (the host polling device 12.1, which is probably an input device - the sniffer will see all downstream traffic on the bus, not just the traffic intended for the device behind it). Did you actually do something to the drive, like read/write some data or reconnect it, after starting to sniff?

In order to avoid noise from other devices on the bus, I would recommend connecting the sniffer's sniff-side upstream port to a USB controller with no other devices connected, though that is not a requirement.

marekmatw, 2018-09-11:
I do no operations when sniffing. There is just one file with an `*.txt` extension on it.

I also have no devices connected. Just the OV. Maybe it is like that becaues I use a drive USB3.0

marcan, 2018-09-11:
If you do nothing to the drive while sniffing then of course you will get no traffic. Try doing something?

OpenVizsla does not support USB3.0, but any SuperSpeed devices should fall back to High Speed mode when connected through OV.

marekmatw, 2018-09-11:
I noticed that sniffing stops in one point. For instance

```
PERR: 0002 (Overflow)
[      O ]  11.881343 d=  0.000000 [ 87   +128.433] [196] DATA1: 25 e5 13 f3 ae 2e 88 ba 87 b6 52 df 5f df 56 70 23 cd 57 b9 8a 68 e6 f2 1d 35 0d b5 c6 e0 59 4e 65 de d1 af 04 84 c9 e5 16 a0 86 0e a5 12 ff a1 4b ca 7a 92 7d db eb f6 5c b1 15 92 52 7d e8 2a 5c 05 b5 7c 99 ab a5 50 a7 dc 3e b6 85 eb 94 f2 00 83 e2 ed 97 e1 b9 b7 53 1f 6a 0e 02 80 08 30 1d d7 0f c1 77 47 91 03 05 05 5e fe 80 c7 6a 85 45 82 f9 2b f5 56 e1 4d ae 5a e9 4f 22 e0 58 f7 52 b0 9f 3f 3e 7e b8 4e f1 52 4e b7 88 43 ac 13 54 d6 24 48 43 d0 09 2f a0 78 2f 1e 3e 8f 2e f6 34 f8 bc 49 0d d2 02 97 77 b8 8e fa 46 da 3c b1 8c 82 02 d5 a9 b3 e4 67 3b 2e c7 61 4b e0 95 0c d7 64 29	Unexpected ERR CRC 
[        ]  11.881347 d=  0.000004 [ 87   +131.967] [ 53] Err - bad PID of 0f 
[        ]  11.881348 d=  0.000001 [ 87   +133.117] [  1] ACK 
[        ]  11.881350 d=  0.000003 [ 87   +135.883] [467] Err - bad PID of 0c 
[        ]  11.881359 d=  0.000008 [ 87   +143.967] [  1] ACK 
[        ]  11.881359 d=  0.000001 [ 87   +144.733] [  3] IN   : 31.1 
[        ]  11.881369 d=  0.000009 [ 87   +154.200] [  1] ACK 
[        ]  11.881369 d=  0.000000 [ 87   +154.683] [  3] OUT  : 33.2 
```

marcan, 2018-09-11:
Overflow suggests the OV can't keep up with the throughput on the bus. If you have it on the same controller as the target device, you should move it to a different controller to maximize your available bandwidth. Also keep in mind you cannot sniff a bus that is fully utilized forever, since the available sniffing throughput is less since it has to itself go over USB with overhead.

ar0mat, 2018-10-09:
Everything works. Thank you @marcan for your help and time.

closed 2018-10-09 by marcan