openvizsla/ov_ftdi #19: EEPROM programming issue

ar0mat, 2018-10-09:
Hello!
I'm trying to program the EEPROM. I have been stuck in this step.
```
$ sudo python3 ovctl.py -p ov3.fwpkg eep-program 10000
FPGA not loaded, forcing reload
FPGA: Bitstream timestamp 2014/11/10 16:31:07
FPGA: sending configuration bitstream
FPGA: CRC OK
FPGA: configured
00: ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff 
10: ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff 
20: ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff 
30: ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff(ffff)
EEPROM: Device blank or checksum incorrect
EEPROM: Generated serial OV010000
```
After the assembly process, I noticed that EEPROM had been soldered other way around. I have that chip replaced by a new one, but it still does not work. 

FTDI and Y2 oscillator are soldered correctly.

marcan, 2018-10-09:
What does eep-program say after you fixed the EEPROM? Does it complain or do you get a successful `EEPROM: Programmed` message?

If programming fails then you might still have a hardware issue. Check for short and open pins.

ar0mat, 2018-11-21:
Here is what I receive after fixing the EEPROM:
```
$ sudo python3 ovctl.py -p ov3.fwpkg eep-program 1421312
00: ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff
10: ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff
20: ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff
30: ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff(ffff)
EEPROM: Device blank or checksum incorrect
EEPROM: Generated serial OV142131
```

Basically, nothing has changed. Unplugging hasn't solved the problem. I've checked the connections. There is no fault in assembly. When I enumerate devices with `lsusb` command, I receive
```
$ lsusb
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 005: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 001 Device 004: ID 046d:c077 Logitech, Inc. M105 Optical Mouse
Bus 001 Device 011: ID 0403:6010 Future Technology Devices International, Ltd FT2232C Dual USB-UART/FIFO IC
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
so FT2232H is recognizable. 

I'm curious what are these lines responsible for:
```
00: ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff
10: ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff
20: ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff
30: ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff ffff(ffff)
```

marcan, 2018-11-21:
If you don't get an "EEPROM: Programmed" message on `eep-program` then programming is failing. Check your FTDI and EEPROM pins for shorts and opens. If the EEPROM has been replaced then maybe something on the FTDI is broken, though honestly I would be surprised if placing the EEPROM on backwards would damage the FTDI.