openvizsla/ov_ftdi #22: Burn the firmware step file

wxh0000mm, 2018-12-04:
Hello:
OV3 folder has been built. I have not found out how to burn the built firmware into the chip. Could you please provide me with the corresponding files?

wxh0000mm, 2018-12-04:
First contact, there are many places do not understand, please give me guidance.thank you

somehdlguy, 2018-12-04:
The FPGA bitstream isn't burnt into the chip or board. It is loaded on-demand when you use `ovctl.py` to talk to the board.
The only programmable part of the board is the FTDI EEPROM. You shouldn't need to do this with a board you got from the kickstarter! It should already be programmed. If you have a hand-built board, or if you somehow erased the EEPROM, you can reprogram the FPGA with:

- `ovctl.py eep-erase`
- `ovctl.py eep-program SERIALNO`

wxh0000mm, 2018-12-05:
Unmatched byte 10 - discarding
Unmatched byte 08 - discarding
Unmatched byte 10 - discarding
Unmatched byte 08 - discarding
Unmatched byte 10 - discarding
Unmatched byte 18 - discarding
Unmatched byte 10 - discarding
Unmatched byte 18 - discarding
Unmatched byte 10 - discarding
Unmatched byte 18 - discarding
Unmatched byte 10 - discarding
Unmatched byte 18 - discarding
Unmatched byte 10 - discarding
Unmatched byte 00 - discarding
Unmatched byte 10 - discarding
Unmatched byte 00 - discarding
Unmatched byte 10 - discarding
Unmatched byte 00 - discarding
Unmatched byte 10 - discarding
Unmatched byte 00 - discarding
0 / 16777216 (0.00 % utilization) 144 kB | 0 overflow, 4e4bc6e9 total | R0002402e W0002402e
0 overflow, 52454720 total
Unmatched byte 10 - discarding
Unmatched byte 08 - discarding
Unmatched byte 10 - discarding
Unmatched byte 18 - discarding
Unmatched byte 10 - discarding
Unmatched byte 18 - discarding
Unmatched byte 10 - discarding
Unmatched byte 18 - discarding
Unmatched byte 10 - discarding
Unmatched byte 18 - discarding
Unmatched byte 10 - discarding
Unmatched byte 00 - discarding
Unmatched byte 10 - discarding
Unmatched byte 00 - discarding
Unmatched byte 10 - discarding
Unmatched byte 00 - discarding
Unmatched byte 10 - discarding
Unmatched byte 00 - discarding

hello
I use sudo. / ovctl.py-p ov3wpkg sniffing ls command and the information is exposed like the above. Why?

wxh0000mm, 2018-12-05:
@somehdlguy 
king@king-virtual-machine:~/桌面/ov_ftdi/software/host$ sudo ./ovctl.py eep-program serialno
usage: ovctl.py eep-program [-h] serialno
ovctl.py eep-program: error: argument serialno: invalid int value: 'serialno'

My is a manual circuit board, can you complete the following two commands?
Ovctl. Py eep - erase
Ovctl. Py eep - program SERIALNO

closed 2018-12-05 by wxh0000mm