openvizsla/ov_ftdi #25: Error when building software for FPGA

ar0mat, 2019-03-25:
Hello everyone.

I have been trying to build "firmware" for an FPGA and I'm receiving such error

```
Traceback (most recent call last):
  File "build.py", line 28, in <module>
    top = OV3(plat)
  File "/home/user/projects/ov_ftdi/software/fpga/ov3/ovhw/top.py", line 114, in __init__
    ftdi_io)
  File "/home/aromat/projects/3mdeb/openvizsla_new/ov_ftdi/software/fpga/ov3/ovhw/ftdi_bus.py", line 136, in __init__
    self.specials += MultiReg(~io.wr_n, self.tx_ind)
NameError: name 'MultiReg' is not defined
Makefile:18: recipe for target 'build/ov3.bit' failed
make: *** [build/ov3.bit] Error 1
```

When I used this soft 4 months ago, I didn't have errors like this.

artur-rs, 2019-04-02:
I have found that `software/fpga/ov3/ovhw/ftdi_bus.py` file is missing crucial import line:
```
from migen.genlib.cdc import MultiReg
```

Simply add above line in the mentioned file and it should work as expected.

pietrushnic, 2019-04-02:
@artur-rs it would be great if you can provide PR that fix the problem :)