openvizsla/ov_ftdi #32: Err - bad PID of 0f

attie, 2019-08-14:
I've just received my OpenVisla v3.2 from sysmocom (Thanks!), and I'm seeing some unexpected output.

With no downstream device connected (J1 and J2 are connected to the same host, J3 is not connected to anything) I see the following...

```
$ ./software/host/ovctl.py sniff fs
FPGA not loaded, forcing reload
FPGA: Bitstream timestamp 2014/11/10 16:31:07
FPGA: sending configuration bitstream
FPGA: CRC OK
FPGA: configured
[   F    ]   0.178034 d=  0.178034 [   .0 +178033.667] [  0]
PERR: 0001 (Error)
[       E]   0.229603 d=  0.051569 [   .0 +229603.150] [  0]
[        ]   0.229604 d=  0.000001 [   .0 +229604.083] [  1] Err - bad PID of 0f
PERR: 0001 (Error)
[       E]   0.238475 d=  0.008871 [   .0 +238474.900] [  0]
[        ]   0.238476 d=  0.000001 [   .0 +238475.833] [  1] Err - bad PID of 0f
PERR: 0001 (Error)
[       E]   0.239591 d=  0.001115 [   .0 +239591.317] [  0]
[        ]   0.239592 d=  0.000001 [   .0 +239592.250] [  1] Err - bad PID of 0f
PERR: 0001 (Error)
[       E]   0.240150 d=  0.000558 [   .0 +240150.067] [  1] Err - bad PID of 0a
[        ]   0.240152 d=  0.000002 [   .0 +240151.667] [  1] Err - bad PID of 0f
PERR: 0001 (Error)
[       E]   0.242733 d=  0.002582 [   .0 +242733.317] [  0]
[        ]   0.242734 d=  0.000001 [   .0 +242734.250] [  1] Err - bad PID of 0f
PERR: 0001 (Error)
[       E]   0.257584 d=  0.014849 [   .0 +257583.567] [  0]
[        ]   0.257584 d=  0.000001 [   .0 +257584.500] [  1] Err - bad PID of 0f

----- >8 -----

PERR: 0001 (Error)
[       E]  33.044570 d=  0.045737 [   .0 +33044569.633] [  0]
[        ]  33.044571 d=  0.000001 [   .0 +33044570.567] [  1] Err - bad PID of 0f
PERR: 0001 (Error)
[       E]  33.084541 d=  0.039970 [   .0 +33084540.717] [  1] Err - bad PID of 0c
[        ]  33.084542 d=  0.000002 [   .0 +33084542.317] [  1] Err - bad PID of 0f
0 / 16777216 (0.00 % utilization) 21 kB | 0 overflow, ffffffff total | R00005748 W00005748
0 overflow, ffffffff total
0 / 16777216 (0.00 % utilization) 21 kB | 0 overflow, ffffffff total | R00005748 W00005748
0 overflow, ffffffff total
0 / 16777216 (0.00 % utilization) 21 kB | 0 overflow, ffffffff total | R00005748 W00005748
0 overflow, ffffffff total
0 / 16777216 (0.00 % utilization) 21 kB | 0 overflow, ffffffff total | R00005750 W00005750
0 overflow, ffffffff total
0 / 16777216 (0.00 % utilization) 21 kB | 0 overflow, ffffffff total | R00005759 W00005759
0 overflow, ffffffff total
PERR: 0001 (Error)
[       E]  33.207216 d=  0.122674 [   .0 +33207216.300] [  0]
[        ]  33.207217 d=  0.000001 [   .0 +33207217.233] [  1] Err - bad PID of 0f
PERR: 0001 (Error)
[       E]  33.438442 d=  0.231225 [   .0 +33438441.783] [  0]
[        ]  33.438443 d=  0.000001 [   .0 +33438442.717] [  1] Err - bad PID of 0f
PERR: 0001 (Error)
[       E]  33.525680 d=  0.087238 [   .0 +33525680.317] [  0]
[        ]  33.525681 d=  0.000001 [   .0 +33525681.250] [  1] Err - bad PID of 0f
PERR: 0001 (Error)
[       E]  33.618852 d=  0.093171 [   .0 +33618852.467] [  0]
0 / 16777216 (0.00 % utilization) 21 kB | 0 overflow, ffffffff total | R00005761 W00005761
0 overflow, ffffffff total
0 / 16777216 (0.00 % utilization) 21 kB | 0 overflow, ffffffff total | R0000576a W0000576a
0 overflow, ffffffff total
0 / 16777216 (0.00 % utilization) 21 kB | 0 overflow, ffffffff total | R0000576a W0000576a
0 overflow, ffffffff total
```

Is this a known issue or something that's been observed before?
It's connected to the front ports of a NUC (NUC8i7BEH).

With just the host connected to J1 (disconnected from J2), things appear to be okay:

```
venv:$ ./software/host/ovctl.py sniff fs
FPGA not loaded, forcing reload
FPGA: Bitstream timestamp 2014/11/10 16:31:07
FPGA: sending configuration bitstream
FPGA: CRC OK
FPGA: configured
0 / 16777216 (0.00 % utilization) 0 kB | 0 overflow, 00c33d5f total | R00000004 W00000004
0 overflow, 02c40b7a total
0 / 16777216 (0.00 % utilization) 0 kB | 0 overflow, 09a1cdd3 total | R00000004 W00000004
0 overflow, 0ba29678 total
0 / 16777216 (0.00 % utilization) 0 kB | 0 overflow, 12803617 total | R00000004 W00000004
0 overflow, 1480ee26 total
0 / 16777216 (0.00 % utilization) 0 kB | 0 overflow, 1b5e9ff9 total | R00000004 W00000004
0 overflow, 1d5f5ab2 total
0 / 16777216 (0.00 % utilization) 0 kB | 0 overflow, 243d2197 total | R00000004 W00000004
0 overflow, 263de1f5 total
0 / 16777216 (0.00 % utilization) 0 kB | 0 overflow, 2d1ba392 total | R00000004 W00000004
0 overflow, 2f1c278b total
0 / 16777216 (0.00 % utilization) 0 kB | 0 overflow, 35fa0a6a total | R00000004 W00000004
0 overflow, 37faaa40 total
0 / 16777216 (0.00 % utilization) 0 kB | 0 overflow, 3ed87628 total | R00000004 W00000004
0 overflow, 40d90a39 total
0 / 16777216 (0.00 % utilization) 0 kB | 0 overflow, 47b6c63e total | R00000004 W00000004
0 overflow, 49b7871b total
0 / 16777216 (0.00 % utilization) 0 kB | 0 overflow, 50953fba total | R00000004 W00000004
0 overflow, 52960368 total
```

---

Some info:

```
$ md5sum software/host/ov3.fwpkg
ea64fd5dcfa0ca710e4b22fd2da42738  software/host/ov3.fwpkg
$ git rev-parse HEAD
61ddaf4afdae046b13e24584fdc59af0fc141ba5
```
```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.2 LTS
Release:        18.04
Codename:       bionic
$ apt show libusb-1.0-0
Package: libusb-1.0-0
Version: 2:1.0.21-2
Priority: standard
Section: libs
Source: libusb-1.0
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Aurelien Jarno <aurel32@debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 127 kB
Depends: libc6 (>= 2.17), libudev1 (>= 183)
Homepage: http://www.libusb.info
Task: standard
Supported: 5y
Download-Size: 43.3 kB
APT-Manual-Installed: yes
APT-Sources: http://gb.archive.ubuntu.com/ubuntu bionic/main amd64 Packages
Description: userspace USB programming library
 Library for programming USB applications without the knowledge of Linux
 kernel internals.
 .
 This package contains what you need to run programs that use this library.
```

I also had to apply [this patch](https://github.com/attie/ov_ftdi/commit/6171077bf4812ea2ed48a0e069d603c3ace25529) to build it with Ubuntu 18.04's packaged libusb...

somehdlguy, 2019-08-14:
I'm not sure what you are seeing is actually a problem - it looks like that might just be noise or truncated packets on the line. The FPGA is passing back packets received from the USB bus, but they don't have a sane start PID. The OV tries to report exactly what it sees on the line, including corrupt or invalid traffic that would be ignored by a real device.

If you attach a  USB device, do you get good captures/traffic?

attie, 2019-08-15:
Thanks for your response.

I hadn't reviewed the schematic in detail yet, but given the (probably) high impedance of the signals, some garbage might be expected - I don't expect `targetDtap_P` (FPGA pin 105) or `targetDtap_N` (FPGA pin 104) will sink / source current...?

> it looks like that might just be noise or truncated packets on the line
> If you attach a USB device, do you get good captures/traffic?

I think I agree with you - captures appear to be correct, and I don't **remember** seeing the errors with a device connected... I'll look out for them and confirm later.

somehdlguy, 2019-08-15:
targetDtap_P and _N should be high-Z. The entire OV USB path is high-Z and relies on the device/host to do bus termination. (we configure the PHY to leave internal termination off; and the stub length is short enough that reflections shouldn't be _too_ bad).

So, it could be noise on an unterminated high-Z bus. I've also seen hubs leak packets or packet fragments that are destined to other downstream ports; so you might be seeing something like that.

tmbinc, 2019-08-15:
Also keep in mind that we're not auto-sensing speed. So if you're
listening to data at the wrong speed, you likely get interesting things
as well.

attie, 2019-08-19:
> I've also seen hubs leak packets or packet fragments that are destined to other downstream ports; so you might be seeing something like that.

Shiny!

> So if you're listening to data at the wrong speed, you likely get interesting things as well.

Understood.

Thanks very much both.

closed 2019-08-19 by attie