openvizsla/ov_ftdi #36: FPGA Models: XC6SLX9-2TQG144C vs. XC6SLX9-3TQG144C (speed)

rljacobson, 2019-10-30:
The BOM specifies an `XC6SLX9-3TQG144C` FPGA. The `3` after the `-` is the "speed grade" [according to the datasheet](https://www.xilinx.com/support/documentation/data_sheets/ds160.pdf). I can buy four `XC6SLX9-2TQG144C`s (speed grade 2) for the cost of one `XC6SLX9-3TQG144C`.

Can anyone tell me if the `XC6SLX9-2TQG144C` is fast enough for and compatible with this application?

This isn't a tool I necessarily need. I got the [PCB for 1€ at sysmocom.de](http://shop.sysmocom.de/t/usb/tracer-analyzer) and plan to assemble it bit by bit as I acquire the parts for the experience. (If anyone else is reading this, sysmocom.de is selling the kit for 47€, which is a bargain, IMHO.)

LoveMHz, 2020-01-31:
I also would be interested in knowing if the XC6SLX9-2TQG144C is a suitable substitute. I'm not familiar enough with the project to discern if it would work, but the [DC and Switching Characteristic datasheet](https://www.xilinx.com/support/documentation/data_sheets/ds162.pdf) has more information on the speed grade differences.

laf0rge, 2021-03-11:
we (at symocom) have also been wondering if the othe r(more easily sourceable) speed grade is suitable.  However, we just use and sell the boards (and soon will produce more) and have not been involved in the design phase of this project.  I guess somebody will have to try to build the bitstream with the timing parameters of the othe speed grade and actually test...