openvizsla/ov_ftdi #57: Openvizsla sniffing nothing (but transfer amount continues to go up)

jaysonlarose, 2023-04-09:
I'm trying to reliably capture data sent between a Nintendo Switch Pro Controller and a Nintendo Switch.

Sometimes this works fine, but more often than not, this sort of situation occurs:

```bash
jayson@grillex % ./ovctl.py sniff fs
Unmatched byte 76 - discarding
Unmatched byte a5 - discarding
Unmatched byte 3f - discarding
Unmatched byte 86 - discarding
0 / 16777216 (0.00 % utilization) 1 kB | 0 overflow, 00c32914 total | R00000713 W00000713
0 overflow, 02c3fb67 total
0 / 16777216 (0.00 % utilization) 15 kB | 0 overflow, 09a1cab5 total | R00003c4d W00003c4d
0 overflow, 0ba2e796 total
0 / 16777216 (0.00 % utilization) 28 kB | 0 overflow, 12803072 total | R0000712c W0000712c
0 overflow, 1480db1b total
0 / 16777216 (0.00 % utilization) 41 kB | 0 overflow, 1b5e99f4 total | R0000a606 W0000a606
0 overflow, 1d5f5e46 total
0 / 16777216 (0.00 % utilization) 54 kB | 0 overflow, 243d0d77 total | R0000dafa W0000dafa
0 overflow, 263de83f total
0 / 16777216 (0.00 % utilization) 67 kB | 0 overflow, 2d1b81df total | R00010fee W00010fee
0 overflow, 2f1c1c82 total
0 / 16777216 (0.00 % utilization) 81 kB | 0 overflow, 35f9d61b total | R000144e2 W000144e2
0 overflow, 37fab6f6 total
0 / 16777216 (0.00 % utilization) 94 kB | 0 overflow, 3ed84eca total | R000179c2 W000179c2
0 overflow, 40d8e51e total
0 / 16777216 (0.00 % utilization) 107 kB | 0 overflow, 47b6c270 total | R0001ae7c W0001ae7c
0 overflow, 49b76b7e total
0 / 16777216 (0.00 % utilization) 120 kB | 0 overflow, 50952b1b total | R0001e35b W0001e35b
0 overflow, 5295cef4 total
0 / 16777216 (0.00 % utilization) 134 kB | 0 overflow, 59739841 total | R00021864 W00021864
0 overflow, 5b748513 total
0 / 16777216 (0.00 % utilization) 147 kB | 0 overflow, 6252efca total | R00024d52 W00024d52
0 overflow, 6452a151 total
0 / 16777216 (0.00 % utilization) 160 kB | 0 overflow, 6b306eb2 total | R00028232 W00028232
0 overflow, 6d312fc4 total
0 / 16777216 (0.00 % utilization) 173 kB | 0 overflow, 740ed717 total | R0002b711 W0002b711
0 overflow, 760f9a02 total
```

The LED next to the FPGA is dimly lit in the same way that it is when sniffing is working, and the kB value is increasing, but no packets are being displayed.

This behavior appears to be "sticky", that is, when sniffing works, chances are ok (if I had to put a number on it, I'd say maybe 75%?) that sniffing will work again if you stop and start the process... but when sniffing doesn't work, chances are very good (I'd say around 90%) that it'll continue to not work.

When it's not working, resetting the USB devices (ie, via the 'usbreset' command) doesn't seem to help any more than just trying again and hoping for the best.

This is only a full speed connection I'm dealing with, so I'd assume this board would have no problems whatsoever dealing with communications between a controller and a games console.  Is there anything I can do to get it to be more reliable?  Or to more fully reset the thing back into a state where I know it'll start sniffing properly when I ask it to do so?

I _am_ using some rather long USB cables to connect between the console and the controller, but as the Switch is having no problems whatsoever picking up what the controller is putting down, I assume this isn't an issue.

Or, occasionally, something like this happens:

```bash
Unmatched byte 03 - discarding
Unmatched byte 00 - discarding
Unmatched byte 7b - discarding
Unmatched byte 8e - discarding
Unmatched byte d6 - discarding
Unmatched byte a5 - discarding
Unmatched byte 24 - discarding
PERR: 00A0 (Last)
Unmatched byte 03 - discarding
Unmatched byte 00 - discarding
Unmatched byte 98 - discarding
Unmatched byte 43 - discarding
Unmatched byte fa - discarding
Unmatched byte a5 - discarding
Unmatched byte 4b - discarding
Unmatched byte 78 - discarding
Traceback (most recent call last):
File "/home/jayson/Git/ov_ftdi/software/host/ovctl.py", line 628, in <module>
main()
File "/home/jayson/Git/ov_ftdi/software/host/ovctl.py", line 621, in main
args.hdlr.go(dev, args)
File "/home/jayson/Git/ov_ftdi/software/host/ovctl.py", line 422, in go
do_sniff(dev, args.speed, args.format, args.out, args.timeout)
File "/home/jayson/Git/ov_ftdi/software/host/ovctl.py", line 279, in do_sniff
dev.regs.LEDS_MUX_2.wr(0)
File "/home/jayson/Git/ov_ftdi/software/host/LibOV.py", line 210, in wr
self.writefn(self.addr + self.size - 1 - i, (value >> (i * 8)) & 0xFF)
File "/home/jayson/Git/ov_ftdi/software/host/LibOV.py", line 791, in iowrite
return self.io.do_write(self.resolve_addr(addr), value)
File "/home/jayson/Git/ov_ftdi/software/host/LibOV.py", line 346, in do_write
return self.__txn(0x8000 | addr, value, timeout)
File "/home/jayson/Git/ov_ftdi/software/host/LibOV.py", line 363, in __txn
assert r_addr == io_ext
AssertionError
```

And very occasionally, it'll do this:

```bash
jayson@grillex % ./mister_viz_openvizsla.py
Unmatched byte e9 - discarding
Unmatched byte 94 - discarding
Unmatched byte d2 - discarding
PERR: D000 ()
Unmatched byte df - discarding
Unmatched byte aa - discarding
Unmatched byte e1 - discarding
Unmatched byte 02 - discarding
Unmatched byte 81 - discarding
Unmatched byte 01 - discarding
Unmatched byte 40 - discarding
Unmatched byte 40 - discarding
Unmatched byte 85 - discarding
Unmatched byte cd - discarding
Unmatched byte 00 - discarding
Unmatched byte 03 - discarding
Unmatched byte 00 - discarding
Unmatched byte 56 - discarding
Unmatched byte 32 - discarding
Unmatched byte 37 - discarding
Unmatched byte a5 - discarding
Unmatched byte 91 - discarding
Unmatched byte 1c - discarding
Unmatched byte 94 - discarding
Unmatched byte dc - discarding
Unmatched byte 3d - discarding
Unmatched byte 5a - discarding
Unmatched byte 9e - discarding
Unmatched byte 14 - discarding
Unmatched byte 0e - discarding
Unmatched byte c5 - discarding
Unmatched byte a1 - discarding
Unmatched byte 74 - discarding
Unmatched byte a7 - discarding
Unmatched byte f4 - discarding
Unmatched byte ff - discarding
Unmatched byte da - discarding
Unmatched byte 00 - discarding
Unmatched byte f4 - discarding
Unmatched byte 07 - discarding
Unmatched byte 80 - discarding
Unmatched byte f0 - discarding
Unmatched byte 87 - discarding
Unmatched byte 80 - discarding
Unmatched byte 0b - discarding
Unmatched byte 72 - discarding
Unmatched byte 03 - discarding
Unmatched byte 2d - discarding
Unmatched byte 00 - discarding
Unmatched byte 17 - discarding
Unmatched byte 10 - discarding
Unmatched byte fc - discarding
Unmatched byte ff - discarding
Unmatched byte f9 - discarding
Unmatched byte ff - discarding
Unmatched byte 00 - discarding
Unmatched byte 00 - discarding
Unmatched byte 70 - discarding
Unmatched byte 03 - discarding
Unmatched byte 1c - discarding
Unmatched byte 00 - discarding
Unmatched byte 2e - discarding
Unmatched byte 10 - discarding
Unmatched byte fb - discarding
Unmatched byte ff - discarding
Unmatched byte fb - discarding
Unmatched byte ff - discarding
Unmatched byte 00 - discarding
Unmatched byte 00 - discarding
Unmatched byte 97 - discarding
Unmatched byte 03 - discarding
Unmatched byte 3f - discarding
Unmatched byte f5 - discarding
Unmatched byte ff - discarding
Unmatched byte fd - discarding
Unmatched byte ff - discarding
PERR: 4040 ()
Unmatched byte 00 - discarding
Unmatched byte 00 - discarding
Unmatched byte 00 - discarding
Unmatched byte 00 - discarding
Unmatched byte 00 - discarding
Unmatched byte 00 - discarding
Unmatched byte 00 - discarding
Unmatched byte d6 - discarding
Unmatched byte a9 - discarding
( at this point, the LED blinking stops, as if the viszla has given up )
```

desowin, 2023-04-09:
Do you observe the same with latest software from https://github.com/OpenVizslaTNG/ov_ftdi ?

How do you connect the Pro Controller with Switch via OpenVizlsa? What type of cables/usb-c adapters do you use?

desowin, 2023-04-09:
To force fpga gateware reset, just add `--load` before `sniff`.

jaysonlarose, 2023-04-09:
Hi! Thanks for responding!

I didn't know anything about  this OpenvizslaTNG repository! I'll have to give it a try!

I found (only in the past 60 minutes or so) that making sure I always call ovctl with the `-l` flag so that it reloads the FPGA gateware seems to give me good results.  I'm reliably able to see the data transfer between controller and console.

I'm using no-name USB cables, nothing real special.

The Switch -&gt; openvizsla cable is a monoprice cable that I bought maybe 8 years ago, it's about 10 feet long and I've never used it until now. 

The openvizsla -&gt; pro controller cable is this weird USB A to C cable that has magnets embedded periodically throughout its length.... it's branded "Supercalla".

The openvizsla -&gt; analysis computer cable is another monoprice cable I've had since forever.

The printing on both of the Monoprice cables says "USB2.0 SHIELDED HIGH SPEED CABLE"

jaysonlarose, 2023-04-09:
Yes, exactly! Using `--load` is the magic bullet that basically solved my issues.  Now, onto the fun stuff :)

jaysonlarose, 2023-04-09:
(closing issue as "completed")